arduino/Arduino #7208: cannot close an unsaved sketch window after edit / crash

JAY-M-ERICSSON, 2018-02-11:
Arduino 1.8.5 on Mac OS X 10.13.3 (High Sierra)

How to reproduce:
- launch the IDE
- open a new sketch --> window N°1
- open a second sketch --> window N°2
- type any text in the second window N°2 so that the sketch is marked as modified
- press cmd-w to close the second window
   --> you get a modal dialog panel stating "Do you want to save changes to this sketch before closing"
- answer "don't save"
--> RESULT: the alert panel get discarded but the window N°2 is still opened

- press cmd-w again  to close the second window
--> RESULT: the IDE crashes

facchinm, 2018-02-12:
Hi @JAY-M-ERICSSON ,would you mind testing if this bug also applies to the Beta build (https://www.arduino.cc/en/Main/Software#beta) ? Thanks!

technoblogy, 2018-02-12:
Arduino 1.8.5 on Mac OS X 10.13.3 (High Sierra)

I can confirm that I am getting this same behaviour of a window staying open after trying to close it, followed by a crash.

Also, I frequently get a crash when opening a sketch from the **Sketchbook** submenu, or creating a new sketch with **New**.

Also, the **Open Recent** submenu is sometimes unexpectedly empty.

The problem still occurs on the Beta in the link in the previous comment.

00alis, 2018-02-13:
Just tested on High Sierra 10.13.3 and Arduino 1.8.5 and everything works fine for me.

@JAY-M-ERICSSON 
> answer "don't save"
--> RESULT: the alert panel get discarded but the window N°2 is still opened

In your steps to reproduce in here windows N 1 is still open, not the N 2. Then you try to close the window N 1 by pressing CMD+W, could you confirm?

technoblogy, 2018-02-13:
For me the problem is very erratic. Sometimes I don't encounter a problem. Another day I might get a crash every time I try to open a sketch.

JAY-M-ERICSSON, 2018-02-13:
When I do close from the window / close menu it works 

the challenge seems to come from  the cmd-W  - the window #2 does not close and a second cmd-W crashes 

If you get it to work once using the menu instead of the shortcut then the bug does not appear and subsequent trials seems to not crash...

JM

JAY-M-ERICSSON, 2018-02-22:
Hey

just tested with PR-beta1.9-BUILD-37 and the problem is gone.

ubidefeo, 2019-04-05:
I'm going to freshen this up :)
issue is still present in 1.8.9

per1234, 2019-04-05:
@ubidefeo the discussion seems to indicate that it was resolved in the beta build of the Arduino IDE, but there is no mention of whether the fix was also in the `master` branch. I don't have an OSX system to try it out. Are you able to reproduce this when using the beta build?:
https://www.arduino.cc/en/Main/Software#beta
If so, I'll re-open the issue.

ubidefeo, 2019-04-05:
@per1234 
I haven't had this issue with 1.9 Beta

technoblogy, 2019-04-14:
Please reopen this issue. I've just had a crash with 1.8.9 (release version) after attempting to open an old sketch from the **Sketchbook** menu, as described earlier in this thread. After restarting the Arduino app the sketch was opened.

per1234, 2019-04-14:
Although the issue could technically be considered fixed, I remember now that facchinm told me the procedure is to leave issues that are only fixed in the `ide-1.9.x-beta` branch open until the fix is merged back into the `master` branch.

So I've added the `IDE 1.9.x Beta` label so that we can more easily find this after the merge and reopened it.