arduino/Arduino #7212: Library inclusion request: FlightSimSwitches

jbliesener, 2018-02-12:
Please include the library at https://github.com/jbliesener/FlightSimSwitches with the library manager.

per1234, 2018-02-12:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

jbliesener, 2018-02-13:
Done.

cmaglie, 2018-02-13:
Library added.

closed 2018-02-13 by cmaglie

jbliesener, 2018-02-18:
Hmmm, I've updated to version 1.0.5 on Github and tagged (released) it last night. Arduino library manager still shows 1.0.3 and does not update. I understand that the index is rebuilt every hour. Anything wrong with my setup?

per1234, 2018-02-18:
@jbliesener I don't see anything wrong with your tag. There is another report of the same problem (https://github.com/arduino/Arduino/issues/7238) so I think the problem is with the indexer job. We'll track further progress on this topic in #7238.