arduino/Arduino #7213: Arduino IDE 1.8.5: unable to load java runtime environment

Loveboeing, 2018-02-13:
[image]

facchinm, 2018-02-13:
@Loveboeing we need FAR more information to help debugging your issue. Which OSX version are you running? Do you have any particular Gatekeeper configuration?

per1234, 2018-07-26:
@Loveboeing we're still waiting to hear back from you regarding facchinm's questions. Or if you have since resolved the problem please take a minute to leave a note here.

per1234, 2018-09-26:
Closing due to lack of response.

@Loveboeing if you'll provide the requested information I'll be happy to re-open this issue.

closed 2018-09-26 by per1234