arduino/Arduino #7217: Changed Arduino library name. Please update index.

LowPowerLab, 2018-02-14:
https://github.com/LowPowerLab/SPIFlash

Due to a naming conflict and no action from the creator of the offending library (https://github.com/Marzogh/SPIFlash), I renamed my library to `SPIFlash_LowPowerLab`. I created the library.properties file in the master branch with these changes:
https://github.com/LowPowerLab/SPIFlash/blob/master/library.properties

Please update the library manager/index.

Thank you!

per1234, 2018-02-14:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

LowPowerLab, 2018-02-14:
How about this one:
https://github.com/LowPowerLab/SPIFlash/releases/tag/101.0.0

cmaglie, 2018-02-20:
done

closed 2018-02-20 by cmaglie