arduino/Arduino #7220: New library version not being shown

feikname, 2018-02-14:
I'm having the same problem with this library of mine: https://github.com/Spaguetron/ST_HW_HC_SR04

I updated the tags to a new scheme (where `version` matches the tag names now), but it seems 2.0.3 is not being added to the Library Manager.

facchinm, 2018-02-16:
Hi @feikname ,
in fact it is due to the library name change; the library crawler expects `ST_HW_ST_HCRS04` as name, and changing the github URL will probably remove older releases for the same reason. @cmaglie probably knows how to handle it in a better way

feikname, 2018-02-16:
Just to clarify, I only updated the `name` field (which was accidentally not matching the name from the GitHub repo). The URL hasn't changed.

cmaglie, 2018-02-20:
I've updated the name of the library, now only the latest version is published because it's the only one that has the correct name.

closed 2018-02-20 by cmaglie