arduino/Arduino #7226: Not building Arduino IDE in Eclipse Java Oxygen

soloweyDev, 2018-02-15:
Добрый день,
Скачал из Git версию Arduino IDE. Запускаю ее в Eclipse Java Oxygen. В консоле 3 ошибки. Две из них нехватка библиотек. Что делать?
[image]
Good afternoon,
I downloaded the version of Arduino IDE from Git. I run it in Eclipse Java Oxygen. There are errors in console 3. Two of them lack of libraries. What to do?

facchinm, 2018-02-16:
Hi @soloweyDev ,
the issue should be solved by https://github.com/arduino/Arduino/commit/fa2d18a1a55ecf6f22be6a88ca68b9075a2680d3 , which has been merged in ide-1.9.x-beta branch. Feel free to test it (by switching to beta or cherry-picking the patch) and report if it works

soloweyDev, 2018-02-16:
Thanks, helped.

closed 2018-02-16 by soloweyDev