arduino/Arduino #7227: Library Request for ArduinoArcherPanelClient

mk71, 2018-02-15:
May you please add the library
https://github.com/byteAgenten/ArduinoArcherPanelClient
to the library manager.

Thx
byteAgenten

mk71, 2018-02-20:
Hi - i merged your pull request to the master branch, increased the version to v1.0.1 and created a new tag. Do I have to do anything more?

Kind regards
byteAgenten

per1234, 2018-02-20:
It looks good to me! There's nothing more you need to do. My pull requests were unrelated to your Library Manager index inclusion request. I just take a quick look at libraries when they are posted here so see if there are any improvements I can make. 

cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub at that time and your library will be available in Library Manager shortly after.

cmaglie, 2018-02-20:
done

closed 2018-02-20 by cmaglie