arduino/Arduino #7231: [Library Manager] Add new library request : Gamebuino Meta

Rodot, 2018-02-16:
Please add https://github.com/Rodot/Gamebuino-Meta
Thanks ! 🎮

cmaglie, 2018-02-20:
Looks like this library contains Windows executable files, this is not allowed by our policy

```
JOB 1339 - 2018/02/20 10:21:32 Scraping https://github.com/Gamebuino/Gamebuino-Meta.git
JOB 1339 - 2018/02/20 10:21:46 Checking out tag: 1.0.0
JOB 1339 - 2018/02/20 10:21:46 Error syncing library: *.exe files found, skipping
JOB 1339 - 2018/02/20 10:21:46 Checking out tag: 1.0.1
JOB 1339 - 2018/02/20 10:21:46 Error syncing library: *.exe files found, skipping
```

BTW the library has been added to the indexer, if you remove the .exe and publish a `1.0.2` version it will be published automatically by our crawler, within an hour.
I'm going ahead and close the issue, if you have more questions, please reopen and comment further.

closed 2018-02-20 by cmaglie

Rodot, 2018-02-20:
Thanks a lot ! 👍