arduino/Arduino #7232: IDE fails to start with bad parameter

ggs67, 2018-02-16:
I have installed the Arduino IDE from teh Windows Store on Windows 10 Version 1709 build 16299.248 . When I attempt to start it, it gives only a window with an error message:

```
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86_mdqgnx93n4wtt\arduino.exe

Bad Paremeter

```
Not thet my Windows is a german version and the actual error message is "Falscher parameter" which I translated to "Bad Parameter" or "Wrong parameter"

I tried to deinatll and reinstall with noch chang ein behavior.

matthijskooijman, 2018-02-16:
Can you try starting `arduino_debug.exe` and see if that gives extra output?

feikname, 2018-02-16:
You can't launch Windows Store apps as executables directly IIRC (that would defeat the sandboxing I guess)

See http://www.mrlacey.com/2017/01/launch-store-installed-app-from-command.html?m=1

ggs67, 2018-02-16:
Indeed, don't see how I could do that. I do not even manage to start the app using the start command at all.
This is why I hate Windows Store apps on a PC

matthijskooijman, 2018-02-16:
Ah, not a Windows user myself, so I'm mostly guessing here :-)

AFAIK the regular Windows download also works on Windows 10, so you do not have to use the Store version I believe.

jonaskuske, 2020-04-04:
Apparently you can use/activate modern Windows apps from the command line, need to specify an `appExecutionAlias` in the manifest and then it'll be available under that name: 

https://docs.microsoft.com/en-us/uwp/schemas/appxpackage/uapmanifestschema/element-uap5-appexecutionalias

Described in a Windows blog post here: https://blogs.windows.com/windowsdeveloper/2017/07/05/command-line-activation-universal-windows-apps/

@matthijskooijman is this something you could implement? Would love to see it, the Store version (easy updates and uninstallation..) could be used in combinaton with [vscode-arduino](https://github.com/microsoft/vscode-arduino) then :)

matthijskooijman, 2020-04-04:
@cmaglie, @facchinm, leaving this to you guys. Haven't looked at the links yet, but it sounds like this would be useful for the store version?

facchinm, 2020-04-07:
It would be useful indeed! Testing the windows store app is very hard though, so it's better if someone who already added the property in the right way could patch the manifest https://github.com/arduino/Arduino/blob/master/build/windows/launcher/wrapper-manifest.xml :wink:

jonaskuske, 2020-04-08:
@facchinm Is the build for the Store version documented somewhere? Just followed the steps at https://github.com/arduino/Arduino/wiki/Building-Arduino and it only gives me a `.zip` when compiling for Windows, not a `.appx` or `.msix` app package that I could test.