arduino/Arduino #7238: [Library Manager] Updated library Fabrik2DArduino

henriksod, 2018-02-18:
https://github.com/henriksod/Fabrik2DArduino

New tag 0.6.3

Thank you for your time

per1234, 2018-02-18:
The Library Manager indexer job is supposed to run hourly and automatically pick up any new tag you create so there is no need to create an issue for every release. However, your tags have still not been picked up after 11 hours and I don't see anything wrong with the latest tag so I think something is wrong with the indexer. Further evidence of this is provided by https://github.com/arduino/Arduino/issues/7212#issuecomment-366524095

I'm sure @cmaglie will set it right soon.

per1234, 2018-02-19:
Fabric2D 0.6.3 and FlightSimSwitches 1.0.5 are both now in the Library Manager index so it looks like this issue has been fixed.

closed 2018-02-19 by per1234

cmaglie, 2018-02-20:
It was a network problem on our side, thanks for your patience.