arduino/Arduino #7243: RGBmatrixPanel.h Broken

PreciselyNothing, 2018-02-18:
RGBmatrixPanel.h 

This worked up until the last round of updates.  Just including this definition causes the below error.

Arduino: 1.8.5 (Mac OS X), Board: "Arduino/Genuino Uno"

/Users/xxxx/Documents/Arduino/libraries/RGB_matrixPanel/RGBmatrixPanel.cpp: In member function 'virtual void RGBmatrixPanel::drawPixel(int16_t, int16_t, uint16_t)':
/Users/xxxx/Documents/Arduino/libraries/RGB_matrixPanel/RGBmatrixPanel.cpp:279:14: error: 'swap' was not declared in this scope
     swap(x, y);
              ^
exit status 1
Error compiling for board Arduino/Genuino Uno.

per1234, 2018-02-18:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2018-02-18 by per1234

PreciselyNothing, 2018-02-18:
Who does a broken *.h file, that was previously working not count as a bug?  

-David

Closed #7243.