arduino/Arduino #7247: Separate devices for uploading and for the serial console

probonopd, 2018-02-19:
It would be a good idea to be able to set separate serial devices for uploading and for the serial console in the Arduino IDE. 

In my case, I need `/dev/ttyACM0` for flashing and `/dev/ttyACM1` for the serial monitor. I am using a [nRF5 board](https://github.com/sandeepmistry/arduino-nRF5) with a [Black Magic Probe](https://github.com/blacksphere/blackmagic/wiki) as the programmer.

Reference:
https://github.com/sandeepmistry/arduino-nRF5/issues/250

matthijskooijman, 2019-10-29:
This is very much related to #5554, which essentially requests splitting the device/upload and programmer serial port for the same reason.

per1234, 2022-12-01:
Now tracked at https://github.com/arduino/arduino-ide/issues/583

closed 2022-12-01 by per1234