arduino/Arduino #7248: Add uSevenSegmentLib to Library Manager

Naguissa, 2018-02-20:
Hello,

Following Library Manager FAQ I open this issue to request adding uSevenSegmentLib library to Arduino repository.

Repository URL is: https://github.com/Naguissa/uSevenSegmentLib

This library depends on uTimerLib, which is still in inclusion process: https://github.com/arduino/Arduino/issues/7218

Thanks.

per1234, 2018-02-20:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

Naguissa, 2018-02-20:
Oh, my! What a basic mistake!

Fixed.

Thanks!

cmaglie, 2018-02-20:
done

closed 2018-02-20 by cmaglie