arduino/Arduino #7251: Arduino Windows App Error When Uploading if Using Ctrl+Shift+U shortcut

TheTRCG, 2018-02-21:
It works if upload icon is clicked on but not with shortcut.

matthijskooijman, 2018-02-21:
Upload is Ctrl+U, Ctrl+shift+U is "upload using programmer", I believe. Is that what you intend? If so, please provide some actual information, such as error messages, the board you're using, etc.

TheTRCG, 2018-02-22:
If Ctrl+Shift+U is used it gives the error "An error occurred while uploading the sketch"
However if the button is clicked on using without the shortcut it works fine
As far as I am aware it is only in the windows app i.e from the windows store
Programmer: AVRISP mkll
Board: Arduino Mega

matthijskooijman, 2018-02-22:
So when you hold click and click the button, or select the "Upload using programmer..." it works properly?

Please enable verbose compilation and upload in the preferences, and paste the full output here.

TheTRCG, 2018-02-22:
Only clicking the button works.
Even "Upload Using Programmer" Gives an error

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : stk500v2
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

avrdude done.  Thank you.

An error occurred while uploading the sketch

matthijskooijman, 2018-02-22:
You apparently do not realize that "Upload" (Ctrl+U") and "Upload using programmer" (Ctrl+shift+U) are two different things. You somehow insist on pressing shift, while you are not using a programmer and need the normal "Upload" action.

In short: Press Ctrl-U instead of Ctrl-Shift-U and you get a normal upload.

closed 2018-02-22 by matthijskooijman

TheTRCG, 2018-02-22:
Ah, I understand sorry for wasting your time.
Thank you for your assistance