arduino/Arduino #7253: My lib updates for Joba_Tsl2561 are not reflected for several days. What can I do?

joba-1, 2018-02-22:
Hi,

with #7207 I requested to add my library Joba_Tsl2561 to the library manager.
This was done promptly by @cmaglie  - thank you very much for that.

Since then, I have updated the lib 3 times. In the library manager I can see only the first update while I expected to see all updates within hours as described in the wiki.

I published by creating new releases on github. I don't think I did something different for the releases that do not show up compared to the one that is visible. I checked the lib requirements and can't find an issue.

Here is the releases page, so you can easily see changes I made, and when. I think nothing relevant for the library manager: https://github.com/joba-1/Joba_Tsl2561/releases

Could you please help me making the updates visible?

Thanks & Best Regards,
Joachim

facchinm, 2018-02-22:
Hi @joba-1 ,
the error is caused by a symlink 
```
Error while zipping library: Symlink not allowed: .../github.com/joba-1/Joba_Tsl2561/examples/Autogain/platformio.ini -> ../platformio.ini
```
Once remove, the library manager will catch up and display all the new versions.
Thanks!

closed 2018-02-22 by facchinm

joba-1, 2018-02-22:
ok, thanks
-- 
iphone so i typo