arduino/Arduino #7254: Add flushReceiveBuffer() to HardwareSerial

RobTillaart, 2018-02-22:
In Serial communication sometimes one wants to flush all data in the receive buffer at once. E.g. because data has become invalid. - https://forum.arduino.cc/index.php?topic=530650 - 

The code to do this very fast in HardwareSerial is straightforward,
```
void HardwareSerial::flushReceiveBuffer()
{
  _rx_buffer_head = _rx_buffer_tail;
}
```

PaulStoffregen, 2018-02-22:
Processing has this function, named "clear".

https://processing.org/reference/libraries/serial/Serial_clear_.html

Dunno if following Processing's conventions is still a thing?

RobTillaart, 2018-02-22:
Good point, question 
The Processing Serial.clear(), does that clear the RX buffer or also TX buffer? (no P expert)

If we follow the P convention I propose to add the buffer name so  ```clearReceiveBuffer()``` to make it explicit what is done.

johnholman, 2018-02-22:
To avoid possible race conditions care is needed in the implementation. I've not thought about it carefully, but it might be necessary

1. to add a guard to avoid buffer pointers being changed by the transmission ISR in between reading the ail and assigning to the head.

2. to disable the UDR interrupt otherwise an unfortunately timed interrupt might re-transmit the entire buffer 

Something like this (untested!)

void HardwareSerial::flushReceiveBuffer()
{
 ATOMIC_BLOCK(ATOMIC_RESTORESTATE) {
  _rx_buffer_head = _rx_buffer_tail;
   cbi(*_ucsrb, UDRIE0);
 }

See #6855 for (lengthy) discussion of some of the issues involved, and associated patches #6855 now merged in the AVR core repository.

RobTillaart, 2018-02-22:
(1) makes sense,  however nowhere in the current HardwareSerial (AVR) I see a guard like you mention.

(2) would this be solved by assigning the tail iso the head? a new incoming char would move head 1 position, so at most one char is in the buffer now

```void HardwareSerial::flushReceiveBuffer()
{
_rx_buffer_tail = _rx_buffer_head;  
}```

I will check the discussion (later)

johnholman, 2018-02-22:
Rob  - my earlier message was about flushing the transmit buffer, but later realised that function is already there and you are talking about flushing the receive buffer. So I deleted the message - sorry about the confusion.

A guard might be needed here to prevent the rx_complete_irq function moving the head pointer after read of the head pointer but before write of the tail pointer in this function. I'm not sure whether this would actually cause a problem, but that does need consideration.

If so, something like
  ATOMIC_BLOCK(ATOMIC_RESTORESTATE) {
  _rx_buffer_head = _rx_buffer_tail;
  }
would be safer.

You are right that there are no such guards presently, but there are in the upcoming version in the separate AVR core repository (#6855) to fix a couple of race conditions in the current code.

RobTillaart, 2018-02-22:
Thanks John, (read the discussion)

if we do not use ATOMIC and use 

_rx_buffer_head = _rx_buffer_tail;

**Three cases**  - assuming single byte indices
- if the rx_complete_irq comes before the read of tail -> no problem
- if the rx_complete_irq comes after the read of tail and before write of head, -> head will be overwritten and the new char will effectively be "flushed" too
- if the rx_complete_irq comes after the write of head, -> there is just a new char after the flush (happens all the time.

in case of multi byte indices ATOMIC is needed (no doubt about that)

ATOMIC could be added conditionally depending on ```#if  (SERIAL_RX_BUFFER_SIZE>256).``` but as I don't know how much clock cycles are added by the ATOMIC that might be overkill. Also from maintenance point of view.

So the state of the proposal is now:
```
void HardwareSerial::clearReceiveBuffer()
{
  ATOMIC_BLOCK(ATOMIC_RESTORESTATE) {
    _rx_buffer_head = _rx_buffer_tail;
  }
}
```

johnholman, 2018-02-22:
Drat - I did it again, confusing transmit and receive buffers. But an RX_BUFFER_ATOMIC macro
similar to the TX_BUFFER_ATOMIC macro that applies a guard only when the *receive* buffer is greater than 256 bytes would be suitable.  That  macro doesn't exist yet though, but could also be useful if and when the other large buffer problems are fixed.

So now I'd suggest

// macro to guard critical sections when needed for large RX buffer sizes
#if (SERIAL_RX_BUFFER_SIZE>256)
#define RX_BUFFER_ATOMIC ATOMIC_BLOCK(ATOMIC_RESTORESTATE)
#else
#define RX_BUFFER_ATOMIC
#endif

void HardwareSerial::clearReceiveBuffer()
{
  RX_BUFFER_ATOMIC {
    _rx_buffer_head = _rx_buffer_tail;
  }
}

RobTillaart, 2018-02-22:
There are other places in HWSerial where RX_BUFFER_ATOMIC  should be used too.
I recall from my head  flush() as that does also assign to _rx_buffer_tail / head .

johnholman, 2018-02-22:
There are indeed. Those are addressed in PR #5 in the AVR core repository (which also defines the RX_BUFFER_ATOMIC macro suggested above). That PR hasn't received much interest - perhaps because there would still be race problems for the transmit case that it doesn't address and because large buffers is a corner case that is not fully supported (e.g. you have to edit HardwareSerial.h to implement them).

RobTillaart, 2018-02-22:
All bugs are corner case :)

johnholman, 2018-02-22:
:)