arduino/Arduino #7261: Gamebuino-Classic and Gamebuino-META library update

Rodot, 2018-02-26:
Hello Arduino,

The library Gamebuino is now renamed to Gamebuino-Classic
https://github.com/Gamebuino/Gamebuino-Classic

Also both our libraries Gamebuino-Meta and Gamebuino-Classic don't seem to be updated after several days, even though we tagged it and updated library.properties ?
https://github.com/Gamebuino/Gamebuino-Classic 

Thank you very much and have a great day :)

per1234, 2018-02-26:
In addition to renaming Gamebuino to Gamebuino-classic, you also renamed Gamebuino-Meta to Gamebuino-META:
https://github.com/Gamebuino/Gamebuino-Meta/commit/341bf78893dc3e0b9a38db62949ac1cc90879c53

When you change the `name` value in library.properties the Library Manager indexer entry must be manually updated so that's why the releases after the name change were not automatically added to the Library Manager index.

cmaglie will update the entries for your libraries at the next opportunity.

Rodot, 2018-02-26:
Oh, sorry about that and thank you for the clarification. We're updating everything but now it shouldn't move anymore :)

cmaglie, 2018-03-03:
Done

closed 2018-03-03 by cmaglie