arduino/Arduino #7264: Upload Problem with Arduino Nano

mhmayyan, 2018-02-27:
I am using Arduino Nano with CH340g. When I try to upload, it gives me the following error message: -

```
avrdude: Version 6.3, compiled on Jan 17 2017 at 11:00:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/user/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf"
         User configuration file is "/home/user/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB1
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x52
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x61
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x77
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x72
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x65
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x61
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x64
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x69
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x6e

avrdude done.  Thank you.

An error occurred while uploading the sketch
```

Note that, this problem comes with the latest version of Arduino AVR Boards (1.6.21). However, when I use version (1.6.20) the uploading problem disappears. I am using Ubuntu 16.04.

KavishBhatia, 2018-02-28:
This error occurs due to the updated boot-loader for arduino nano. You should select Atmega328P (Old Bootloader) in Tools -> Processor before uploading the program. 
This resolved the error for me. Hope this helps.

per1234, 2018-02-28:
Thanks KavishBhatia!

@facchinm are Arduino AVR Boards release notes still going to be added to this repository's [revisions.txt](https://github.com/arduino/Arduino/blob/master/build/shared/revisions.txt)? If so, it would be good to add a note about the change to the Nano board definition. I'd be happy to submit a PR for that.

KavishBhatia, 2018-02-28:
I found this on arduino website

[image]

facchinm, 2018-03-02:
@per1234 
since the patch was also applied here it would ok to add it to release.txt :wink:

closed 2018-03-02 by facchinm

awatterott, 2018-03-16:
If you update the bootloader again, I would recommend to use one with automatic baudrate detection.
We are using it since several years and so it doesn't matter which board/baudrate is selected.
* [OptiBoot with Auto-Baudrate+EEPROM Support](https://github.com/svn2github/transistortester/tree/master/bootloaders)
* [OptiBoot with Auto-Baudrate](https://github.com/watterott/Wattuino/tree/master/software/Optiboot)

N0rbert, 2018-08-12:
Got this problem today with old Nano clone with CH340. How should I guess that I should select old bootloader? Spent about an hour checking wiring, USB-cables, other laptops, etc.

It is completely not user-friendly! Do you respect Arduino users?

matthijskooijman, 2018-09-24:
Changing the bootloader baudrate is tricky, since you somehow need the user to select the baud rate (i.e. old/new bootloader). Currently, this is non-intuitive, since for existing users the default silently changes, in order to have things work out of the box for new users (rather, users of new, official, arduinos).

Perhaps it would be interesting to *not* give this option a default value, to force people to make a choice. I suspect this might be useful for more of these additional board options as well (but not all of them, especially the ESP boards that have a ton of option should probably have defaults). To implement this, I guess we need a way to explicitly state the (absence of a) default value in boards.txt. @cmaglie, @facchinm, how would that sound?

myconstructivismmess, 2025-02-13:
I have the same problem with versions 1.6.18, 1.6.20, and the new 1.8.6. It wasn’t happening a few hours ago. I tried reinstalling Arduino and changing AVR Boards versions a couple of times but it didn't work.

Note :
I'm on Linux version 6.13.2-arch1-1.