arduino/Arduino #7281: DFL168A Synchronous version Library

Dafulai, 2018-03-05:
Hello,
  Please add library from gitHub:
https://github.com/Dafulai/DFL168A-OBD2-J1939-J1708-Sync-Library/archive/1.0.0.zip

Thanks

per1234, 2018-03-05:
In order to be added to the Arduino Library Manager index your library must be located in the root of the repository, rather than in the DFL168A_Sync subfolder. 

Please correct the issue, create a new release or tag (or redo the existing release), and then comment here so we know we can proceed with adding your library.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Dafulai, 2018-03-06:
Hello,
  We have changed the link to:
https://github.com/Dafulai/DFL168A-Sync-Library/archive/1.0.0.zip
Thanks

cmaglie, 2018-03-08:
Done!

closed 2018-03-08 by cmaglie