arduino/Arduino #7283: IDE 1.8.5 and cores 1.6.20 repeat offering

BallscrewBob, 2018-03-05:
IDE's 1.8.4 and 1.8.5 keep offering the same AVR package despite it being the (presumed) current 1.6.20.

Probably just needs a small change.

per1234, 2018-03-05:
The latest Arduino AVR Boards version is 1.6.21. The version bundled with Arduino IDE 1.8.4 and 1.8.5 is 1.6.20 so the expected and correct behavior is for the IDE to display updatable boards notifications until you update to Arduino AVR Boards 1.6.21. If you don't want these notifications you can disable them via **File > Preferences > Check for updates on startup**.

BallscrewBob, 2018-03-05:
It is not offering 1,.6.21 as an option.

[image]

BallscrewBob, 2018-03-06:
Thanks Per...It wont let the originator re-open
Tried a re-install after a full wipe of prev IDE. and no change still get the same offering.
This is under Win 7 x64 BTW (should have added that at the get go)
Going to try the  beta as an unzipped to see if it carries through.

BallscrewBob, 2018-03-07:
Whatever the issue is it carries through to 1.9.0 beta run from an unzipped location.