arduino/Arduino #7286: Add RGB Led library

Manghao, 2018-03-05:
Following the instructions provided [here ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ)
[My deposit of libratry to add](https://github.com/Manghao/RGBLed)

Thank you in advance

per1234, 2018-03-05:
Note that since the [`version` property](https://github.com/Manghao/RGBLed/blob/v1.0.1/library.properties#L2) in library.properties was not updated before the 1.0.1 tag was created, only the 1.0.0 version of your library will be available in Library Manager. To fix this you would need to update the `version` property and then redo the tag or create a new one.

Manghao, 2018-03-05:
I forgot to change the version of the file, I just changed that and I pushed the modification

cmaglie, 2018-03-08:
Done!

closed 2018-03-08 by cmaglie