arduino/Arduino #7291: WiFi.scanNetworks() example code a confusion of goals

ReverseEMF, 2018-03-06:
Regarding the Documentation for the:

**WiFi.scanNetworks()**  method 

**url:** [www.arduino.cc/en/Reference/WiFiScanNetworks](https://www.arduino.cc/en/Reference/WiFiScanNetworks) 

A comment in the code is: 

`// if you are connected, print your MAC address:
`

Yet, what follows appears to be an attempt to print a list of SSIDs, but all the subsequent code does is call the WiFi.scanNetworks() method, and then print the return value, which, being of type _byte_, is not likely to produce a list of SSIDs (at least, not directly). 
So:
1. The code claims (in a comment) it is about to access and print the MAC Address -- there is, even, a declaration of a byte array to purportedly capture said MAC Address.
2. Then, in a message sent to the serial pot, one is led to believe an SSID list is on the way.
3. But, what is actually received (via the same serial port) is the number of networks found!

A veritable kerfuffle of goals, claims, and divergent results,  most likely due to a Cut & Paste session left for a never realized future.

SimonePDA, 2018-03-17:
Thank you for pointing out the wrong comment in the example sketch.

It is now fixed.

per1234, 2019-03-26:
Re-opening because the erroneous comment is still present.

Also, note that there are more legitimate issues reported here than that comment.

Here is a suggested replacement for the example code, which will resolve all the issues reported by ReverseEMF:
```c++
#include <SPI.h>
#include <WiFi.h>

char ssid[] = "yourNetwork";  // the name of your network

void setup() {
  Serial.begin(9600);

  int status = WiFi.begin(ssid);

  if (status != WL_CONNECTED) {
    Serial.println("Couldn't get a WiFi connection");
    while (true);
  }
  else {
    // if you are connected, scan for available WiFi networks and print the number discovered:
    Serial.println("** Scan Networks **");
    byte numSsid = WiFi.scanNetworks();

    Serial.print("Number of available WiFi networks discovered:");
    Serial.println(numSsid);
  }
}

void loop() {}
```

SimonePDA, 2019-03-26:
Agreed. the page has now the code suggested.

closed 2019-03-26 by SimonePDA