arduino/Arduino #7305: Bitwise Tutoral - What about XOR

Eddiiie, 2018-03-09:
Hello,
I am reading through the bitwise tutorial to get a better understanding of logical operators and noticed there is no XOR mentioned.    A project I am working on uses XOR heavily to form a checksum on a string of chars for transmission over the wire.

There are probably even more bitwise operators but XOR is pretty important in my life and thought it deserves a shout out.  :)

Thank you.

per1234, 2018-03-09:
Could you post a link to the tutorial you're referring to?

The Arduino Language Reference bitwise operators have a link to the "BitMath Tutorial" in their "See also" section:
http://playground.arduino.cc/Code/BitMath
Which does have XOR.

Eddiiie, 2018-03-09:
Here is the link:   https://www.arduino.cc/en/Tutorial/BitMask

BitMath looks like a better place to learn. :)

cousteaulecommandant, 2018-03-10:
To be fair "bit toggling" is not that common, at least in the context of "bit masks" where you typically want to set or clear bits.  However I'm concerned with the lack of mention of the `~` operator (`&~` is more or less the opposite of `|` and quite used in bit masking operations).

per1234, 2018-03-10:
That's also in http://playground.arduino.cc/Code/BitMath.

It seems like since the BitMath page is more comprehensive than https://www.arduino.cc/en/Tutorial/BitMask and Arduino already has shown acceptance of BitMath by linking to it from the reference pages, that any useful and unique content BitMask should be merged into the BitMath page and the the BitMask page content should be replaced with a link (a redirect if possible) to the BitMath page and any links on arduino.cc to BitMask should be changed to point to BitMath.

The BitMath page is on the Arduino Playground, which can be edited by anyone. There is a password page when you go to edit but the password is already prefilled so you only need to do the captcha and you're in. @Eddiiie or @cousteaulecommandant, would you care to check if any of the BitMask content should be merged to BitMath, and if so go ahead and do the edit?