arduino/Arduino #7307: avandalen

phydeaux99, 2018-03-10:
i bought a weld control - but lost my instructions - i cannot get the web sites to open - i would like to find a copy of your weld controller (and i would be interested in the readout addition shown on youtube)
thanks

per1234, 2018-03-10:
@phydeaux99 please stop spamming the issue tracker with your completely off topic support request for some product we have nothing to do with. It's not going to do you any good and it's very annoying to the hundreds of people who are subscribed to this repository and thus got emails from you.

Related: https://github.com/arduino/Arduino/issues/7306

closed 2018-03-10 by per1234

matthijskooijman, 2018-03-13:
FYI: I just sent an abuse report for this user to Github.

phydeaux99, 2018-03-13:
finei explained what i thought i was doing, and have not repostedi also asked to be removed from the list i had to signup forknock yourself out tough guy