arduino/Arduino #7312: float is rounded (Not precise)

Gabeki17, 2018-03-12:
Problem Description

Float is not even 0,01 precise.
Sketch:

void setup()
{
Serial.begin(115200);
Serial.println("go");
}

float ize = -999999.44;

void loop()
{
Serial.println(ize,10);
delay(1000);
}
Debug Messages

Serial:

go
-999999.4375000000

PaulStoffregen, 2018-03-12:
Of course float isn't precise.  It's well known 32 bit float gives approx 7 decimal digits of precision.  -999999.4 is 7 decimal digits.

As a sanity check, if you compile and run this on a PC, you'll get exactly the same answer.

    #include <stdio.h>
    
    int main()
    {
        float ize = -999999.44;
    
        printf("%f\n", ize);
        return 0;
    }

Gabeki17, 2018-03-13:
Hi Paul,

It is the same for smaller number.

Not precise.

At least not decimal precesion received:

Hi,

Thx.

But still:

void setup()
{
Serial.begin(115200);
Serial.println("go");
}

float ize = -9.44;

void loop()
{
Serial.println(ize,10);
delay(1000);
}

Serial:

go
-9.4399995803

closed 2018-03-13 by facchinm

PaulStoffregen, 2018-03-13:
Not a bug.  The answer is correct, rounding error after 7 digits.

If you run this on a PC, the result is -9.4399995804

#include <stdio.h>

    int main()
    {
        float ize = -9.44;
    
        printf("%.10f\n", ize);
        return 0;
    }

cmaglie, 2018-03-13:
The error between `9.44` and the floating point representation `9.4399995803` is:
`9.44 - 9.4399995803 = 0.00000042`, well less than 7 significant digits.

Of course if you print with:

```
Serial.println(ize,10);
```

you are printing up to the 10th significant digit (11th actually if you consider the integer part `9.xxx`) and you can see the approximation.
When you convert a decimal number to binary you're going to always get this kind of approximation. See also https://www.h-schmidt.net/FloatConverter/IEEE754.html.