arduino/Arduino #7313: Preventing Library Conflicts?

johnyradio, 2018-03-12:
hi, i'm not sure if this is Arduino's fault, or some other. 

I've been unable to compile [this lib ](https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library)on Arduino, as described in [this thread](https://github.com/adafruit/TinyWireM/issues/13). 

Then i scoured my drives for Wire.h, and deleted all instances of the file except those . Also deleted all Digispark files. 

Now i can compile. 

So it seems Arduino looks in several places for libs, and if it finds the same lib in two places, it loads _all_ libs, causing a library conflict. 

Is that correct?

PaulStoffregen, 2018-03-12:
No, that's not correct at all.

If Arduino finds 2 or more copies that could be used (copies for incompatible architectures aren't considered), then it will pick one.  If there's a compile error, or if you've turned on verbose info while compiling in File > Prefs, Arduino will print a message showing you the full paths of every copy it found, and which of those it used.  This message appears in the small black area below your code in the editor.  You might need to grow the size of that area, or scroll up, to actually see the message.

johnyradio, 2018-03-12:
then maybe it picked the 'wrong' one-- one that i deleted. 
how does it pick?

thx

per1234, 2018-03-12:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there. If we do eventually end up determining there is a bug in Arduino's code I'll be happy to reopen this.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2018-03-12 by per1234