arduino/Arduino #7321: Font size change corrupts inactive tabs

steronydh, 2018-03-14:
I have a sketch with several tabs. If I change the font size using the menu, the active tab refreshes correctly. If I then click on another tab, the text isn't drawn correctly and overlaps (see below). If I then change the font scale the active tab refreshes correctly.

[image]

Arduino IDE 1.8.5

facchinm, 2018-03-14:
Confirmed, thanks for reporting, will take a look later today :wink:

facchinm, 2018-03-14:
The bug lies here: https://github.com/bobbylight/RSyntaxTextArea/blame/master/src/main/java/org/fife/ui/rsyntaxtextarea/RSyntaxTextArea.java#L2067

If you modify the font size on the fly and select a tab which hasn't been opened, the result is fine. As soon as you open it once, the repain() doesn't call `refreshFontMetrics()` so the spaces between letters are wrong. Looks quite difficult to fix on our side, probably opening an issue upstream is the way to go.

JaredReabow, 2021-02-17:
This is still not fixed

bobbylight, 2021-08-16:
Howdy - the RSyntaxTextArea source has changed since this ticket was opened.  I'm interested in taking a look again.  Can someone let me know how you're updating the font size on these already-displayed editors and let me know you believe it's failing?

The following two approaches should work, with caveats, provided the editor is displayable:
* Calling `setFont(newFont)` - this works *as long as* only a single font and font size are used for all token types in the editor (using bold & italics is OK).  This updates all token types using the "default" font's font family and size, which by default all tokens use the same font so this just works.
* Use [RSyntaxTextArea.IncreaseFontSizeAction](https://github.com/bobbylight/RSyntaxTextArea/blob/master/RSyntaxTextArea/src/main/java/org/fife/ui/rsyntaxtextarea/RSyntaxTextAreaEditorKit.java#L1322) and [DecreaseFontSizeAction](https://github.com/bobbylight/RSyntaxTextArea/blob/master/RSyntaxTextArea/src/main/java/org/fife/ui/rsyntaxtextarea/RSyntaxTextAreaEditorKit.java#L667), which should always work, even if different fonts are used for different token types.

In any case, let me know how to reproduce this and I'll try to get it fixed in the next RSTA release.

facchinm, 2021-08-16:
Hi @bobbylight ,
thanks for taking a look. I'm attaching a vide of the behaviour (tested with 3.1.3 so the issue is still not fixed).

https://user-images.githubusercontent.com/8843437/129539774-9dfcd287-2db3-49e3-ae3f-8fc79f94ae26.mp4

Let me know if you need any addidional info to reproduce and thanks for your great work!

bobbylight, 2021-08-18:
Is that a standard JTabbedPane - and thus, are the editors in each tab always in the UI until their tab is closed?  Or are you removing RSyntaxTextAreas and adding new ones when tabs are clicked (or something similar like swapping out the Document)?

I'm still struggling a little to see how this can happen - any call to `setFont()` or `setSyntaxScheme()` should refresh the font metrics cache, provided the editor is displayable, which is why I'm wondering if you're not using a standard RSyntaxTextArea-per-tab setup.

If you're doing something like that, one possible fix on my side is to set `metricsNefverRefreshed` to `false` in `removeNotify()`, which would be symmetrics with what we do in `addNotify()` and trigger a refresh of the font metrics cache the next time the editor is added to a UI.

cmaglie, 2021-08-18:
We extended `RSyntaxTextArea` as a `SketchTextArea` here: https://github.com/arduino/Arduino/blob/master/app/src/processing/app/syntax/SketchTextArea.java#L77

We don't swap Document to reuse the same `SketchTextArea`, in the `Editor` we collect a list of `EditorTab` each one having a separate `SketchTextArea`:

https://github.com/arduino/Arduino/blob/master/app/src/processing/app/Editor.java#L122
https://github.com/arduino/Arduino/blob/master/app/src/processing/app/EditorTab.java#L72

> any call to setFont() or setSyntaxScheme() should refresh the font metrics cache, **provided the editor is displayable**

Is there a reason why the editor must be displayable to recalculate the metrics?

cmaglie, 2021-08-18:
Just to give you some more context, here is how we "change tab" in the editor: https://github.com/arduino/Arduino/blob/master/app/src/processing/app/Editor.java#L1437-L1462

maybe it gives some clues maybe not...

cmaglie, 2021-08-18:
...and I think I fixed it by forcing a `setFont` call inside the `selectTab` function: https://github.com/arduino/Arduino/pull/11650

I don't know if this is the correct fix, I see that in `selectTab` the text area is removed from the `JPanel` when is hidden, maybe that is the reason why the font metrics are wrong?

PR #11650 (2021-08-23): Fix: Font size change corrupts inactive tabs
Discussion in https://github.com/arduino/Arduino/issues/7321#issuecomment-899180166 guided the fix.

Fix #7321
commit 240e1b56 (2021-08-23): Force font metrics recalculation when changing tabs

Fix #7321
closed 2021-08-23 by cmaglie