arduino/Arduino #7324: Arduino IDE should recognize and apply settings found in preferences.txt in sketch folders

RichardBronosky, 2018-03-15:
The settings found in the `preferences.txt`file located at [the default locations](https://www.arduino.cc/en/Hacking/Preferences) should be the defaults and the subset of settings found in `$(dirname $CURRENT_INO_FILE)/preferences.txt` should override those values.

Most people don't use a single board. It's way too easy to brick a board when jumping around from one to another. There are so many unique developer workflows out there that we really need to recognize and support them.

There are countless issues created and closed proposing similar features (though most of them suggested using comments in the *.ino file). There is a large demonstrated need for this. No onw has proposed it in an acceptable way. Let's do that here.

A few things we have to decide are:

- What to do about making settings changes via the menus?
- What to do about Arduino IDE's "save preferences on close" behavior?

lmihalkovic, 2018-04-29:
what about the case of having the same code run on multiple board types?!  (a real use case for me now)

qbolsee, 2021-01-19:
This feature would be really helpful. I like the idea of a `preferences.txt` with the same structure as Arduino's `preferences.txt`, but a `.json` file might be more elegant. In any case, this file should be automatically created in the sketch folder and be updated every time the board settings are changed.

If the file is missing or contains nonsense, no problem, we can fallback to the current behavior.

I'm willing to help implement this, anyone else?

closed 2024-05-20 by per1234