arduino/Arduino #7328: i2c hang whole avr - probably twi library loops

smarq8, 2018-03-15:
Why i2c/wire library does not have any timeout fuses? Its known issue froam a years and still no solutions.
Its my 4th project where endTransmission or requestFrom hang my avr.

probably hanging code:
```
  while(TWI_READY != twi_state){
    continue;
  }
or
  while(TWI_MRX == twi_state){
    continue;
  }
or
  while(TWI_READY != twi_state){
    continue;
  }
or
  while(wait && (TWI_MTX == twi_state)){
    continue;
  }
or
  while(TWCR & _BV(TWSTO)){
    continue;
  }
```

In worst case I should receive error or corrupted data so I can repeat operation. Haging whole avr beacuse some random noise on I2C line or damaged I2C device (in some situations) is not acceptet in my opinion. And as I said, many peoples complains on that form a years and still no solutions.

wmarkow, 2018-04-09:
This is a **known issue**
 * arduino/Arduino#1476
 * arduino/Arduino#5481
 * arduino/Arduino#1842
 * arduino/Arduino#2489

There are some proposed solutions but pull request were never merged, so you can cherry-pick or merge them into your own fork of Arduino to use them in your project.

matthijskooijman, 2019-09-06:
I'm closing this issue, in favor of https://github.com/arduino/ArduinoCore-avr/issues/42 (which is in the, now, correct repository).

closed 2019-09-06 by matthijskooijman

ermtl, 2020-03-27:
It's been 9 FUCKING YEARS since that bug was first discussed (2011), countless people pulled their hair trying to understand why their Arduino was freezing, why would it work normally, then suddenly stop, a dozen of times, it's been raised in here, dozens of times people had been told to get lost, use something else, etc ...
#7328
#2418
#1842
#5249
arduino/ArduinoCore-avr#42
Google shows pages after pages of people getting started and who get discouraged by this elusive bug they don't understand. This runs totally contrary of what arduino is supposed to stand for.

If the fix was difficult, if it compromised compatibility or added other problem, it would be understandable, but it's not the case, all that shit is caused by those 2 damn lines of code that obviously can create an infinite loop if for some reason the read operation does not complete :

// wait for read operation to complete
while(TWI_MRX == twi_state){
continue;
}

Yes, I know, this state is not supposed to happen according to the I2C protocol, but guess what ? electrical glitches didn't get the memo.

Countless people, after losing hours or days kind of solved the issue either by making a modified version for themselves, or switching to another library, so there are several implementations of a timeout that are simple and would easily solve the issue.

But arduino developpers stubbornly refuse to fix it !

You think I'm rude ? After 9 years of giving the finger on this issue to the whole community for absolutely no reason, I couldn't care less.

How long before you close it again ?
Just be honest, and put a "fuck you all, WONTFIX" label on it ...