arduino/Arduino #7332: [Library Manager] Please Add PMsensor Library

ekkai, 2018-03-16:
Please add the following to the library manager for "PMsensor" with v1.0release:

https://github.com/ekkai/PMsensor

This library allows an Arduino/Genuino board to sensing Particulate Matter based on the sharp GP2Y1010AU0F

Thanks!

per1234, 2018-03-16:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

ekkai, 2018-03-16:
I've finished. thanks!

cmaglie, 2018-03-16:
Done!

closed 2018-03-16 by cmaglie