arduino/Arduino #7338: WiFi Location library

gmag11, 2018-03-17:
I've developed a library to get location using WiFi APs around with the help of Google maps Geolocation API. It is compatible with mkr1000, esp8266 and esp32.
You can check it on https://github.com/gmag11/WifiLocation

cmaglie, 2018-03-26:
Done!

closed 2018-03-26 by cmaglie

gmag11, 2018-03-27:
I guess script has selected the wrong branch, not master. I've deleted all extra branches. Tags are correctly defined up to last version 1.2.

per1234, 2018-03-27:
The reason that only versions 0.0.1 and 1.2.0 are available from Library Manager is that it uses the `version` field in library.properties for versioning. The `version` value in library.properties for your [v1.0](https://github.com/gmag11/WifiLocation/blob/v1.0/library.properties#L2) and [1.1](https://github.com/gmag11/WifiLocation/blob/1.1/library.properties#L2) releases were both set to 0.0.1 so this caused the 1.1 release to be available as version 0.0.1 in Library Manager and the v1.0 release to not be available at all.

You just need to remember to always update the `version` value before you make a release.