arduino/Arduino #7339: [Suggestion] Add a header defining which board and additional compile settings, the sketch should compile with

timkoers, 2018-03-18:
When dealing with loads of Arduino Sketches, you might get tired of switching the Board from one to another.
A header that auto-selects various build settings would make life a lot easier in this case.

To given an example of such a header
```
/* Arduino board definition header
*
* @arduino.board=Node32s
* @arduino.flashfreq=80MHz
*/
```
I hope others find this an interesting solution too!

per1234, 2018-03-18:
Closing as duplicate of #1481, and many others, including most recently #7324

We welcome bug reports and feature requests but in the future please take the time to search for existing open and closed issues and pull requests before submitting an issue:
https://github.com/arduino/Arduino/issues?&q=
If you do find the issue has already been reported or a fix has been submitted, you are welcome to give it a "thumbs up" by clicking on the smiley face icon at the top right to show your support. Please only comment on the issue if you have additional information to add.

closed 2018-03-18 by per1234

timkoers, 2018-03-20:
Please reopen #1481 and add the label in progress to it.

Sent from Nine<http://www.9folders.com/>

https://github.com/arduino/Arduino/issues?&q=