arduino/Arduino #7346: incorrect platform.txt used when uploading via CLI

Sorunome, 2018-03-19:
I have a custom board for a arduino zero-based thing, `gamebuino:samd:gamebuino_meta_edbg`

Now, uploading via GUI using openocd (programmer) works fine, it uses the platform.txt i added to that correct board path. This is important as the bootloader is larger (0x4000 instead of 0x2000).

However, if i try to upload via CLI it appears to be using the platform.txt file for the normal arduino zero. If i manually edit in there the bootloader size it uploads fine....but it should be using the one i specified.

Here is how i upload via CLI:
```
arduino /tmp/test/test.ino --upload -v --board gamebuino:samd:gamebuino_meta_edbg --useprogrammer
```

facchinm, 2018-03-20:
Hi @Sorunome ,
could you paste the whole verbose output of the given command? I just tried with another out-of-tree board and CLI upload seems to work just fine.
Also, which version of the IDE are you using?

per1234, 2018-07-26:
@Sorunome we're still waiting to hear back from you regarding facchinm's questions. Without that information, it's likely no progress can be made on this issue. Or if you have since found a resolution for your problem, please take a minute to let us know.

per1234, 2018-09-26:
Closing due to lack of response.

@Sorunome if you'll provide the requested information I'll be happy to re-open this issue.

closed 2018-09-26 by per1234