arduino/Arduino #7351: No way to write code onto IDE window screen !

CastersDenis30, 2018-03-20:
I just downladed the last version 1.8.6 and donate 10$ but when I open the software to write my sketch : the place to write code line is gray and it's not possible to enter new code or write something ! What can I do ? Please help ... When I was on previous version 1.6.0 it's working well. Denis

matthijskooijman, 2018-03-20:
Sounds like you have "external editor" enabled in the preferences. Perhaps you enabled in a previous installation?

CastersDenis30, 2018-03-20:
Hello Matthijs,

Many many thanks for your reply and your successfull help !

You were right; I checked Preference and found that external Editor was
'on' ...

Best regards.

Denis

Denis CASTERS
Directeur industriel SCS MDC
d.casters@scsinternationalmali.com
(+223) 82 91 37 20  Bamako
(+33) 6 52 05 77 86 Nimes

Visitez notre site : www.scsinternationalmali.com

facchinm, 2018-03-20:
Closing as solved

closed 2018-03-20 by facchinm

CastersDenis30, 2018-03-21:
> (+223) 82 91 37 20 <+223%2082%2091%2037%2020>  Bamako
> (+33) 6 52 05 77 86 <+33%206%2052%2005%2077%2086> Nimes

Hi again Matthijs,

I don't know if I can ask U something about ARDUINO trouble shooting
without been connected through GitHub but I try anyway ...

Every time I try to download my new sketch or after any code modifications,
I have to declare again the number of communication PORT where my Arduino
IDE card is connected to my laptop. It jumps frequently from COM 3 to COM 4
and COM 5 !!  How would it be possible to fix it once for all ?

Please advise ...

Thanks,
Best regards.

Denis CASTERS
Directeur industriel SCS MDC
d.casters@scsinternationalmali.com
(+223) 82 91 37 20  Bamako
(+33) 6 52 05 77 86 Nimes

Visitez notre site : www.scsinternationalmali.com

2018-03-20 16:55 GMT+00:00 Denis Casters <dcasters@gmail.com>:

matthijskooijman, 2018-03-21:
This issue tracker is only to be used to report bugs or feature requests. This topic (both your original question and the new question) is more appropriate for the Arduino Forum. I'm sure we'll be able to help you over there.

Please do this:

    Read http://forum.arduino.cc/index.php?topic=148850
    If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

FWIW, I don't know the answer to your question myself, sounds like something Windows-specific.