arduino/Arduino #7358: [Library Manager] Please add RARGBLED v1.0.3 to library manager

hippymulehead, 2018-03-21:
URL of Repository: https://github.com/hippymulehead/RARGBLED
Release: https://github.com/hippymulehead/RARGBLED/tree/v1.0.3

per1234, 2018-03-21:
The release must contain a library.properties file but you added that file after creating the 1.0.3 release. Please create a new release that does contain this file.

hippymulehead, 2018-03-21:
URL of Repository: https://github.com/hippymulehead/RARGBLED
Release: https://github.com/hippymulehead/RARGBLED/tree/v1.0.4

per1234, 2018-03-21:
That will work but note that you forgot to update the [`version` ](https://github.com/hippymulehead/RARGBLED/blob/v1.0.4/library.properties#L2)value in library.properties before you made the release. It's that version property that Library Manager uses for versioning, rather than the tag name. So even though the release is v1.0.4, it will be shown as 1.0.3 in Library Manager.

cmaglie, 2018-03-26:
Done!

closed 2018-03-26 by cmaglie