arduino/Arduino #7362: feature request: add board name string to compile defines in platform.txt file

bperrybap, 2018-03-22:
Currently there is a define for the board type -DARDUINO_{build.board} that is passed to the compiler.
But the actual name of the board as a string is not available.
There are some use cases in diagnostic tools where having the actual name of the board as a string can be useful as it can be used to display to the user.
The esp8266 core uses this define -DARDUINO_BOARD="{build.board}" in their platform.txt file.
It would be nice if the Arduino.cc provided cores supported a define that allowed the code to obtain a string version of the board name.
Since the name ARDUINO_BOARD is already used by the esp8266 core, perhaps the Arduino.cc cores could adopt the same thing.

bperrybap, 2018-03-22:
Just a followup. The string in the esp8266 is not really the board name but rather a string version of the build.board value.
What would be nicer for the use case mentioned above would a string version of the actual Arduino board name.

facchinm, 2018-03-23:
Hi @bperrybap ,
I think the issue should be moved to https://github.com/arduino/arduino-builder ; in practice, we don't add any field not specified in the platform.txt , so doing it for everyone could be a problem. 
However, feel free to propose a {build.something} property we can populate so the core interested in exporting such macro (which could be similar to `__BOARDNAME__` ) can do it easily by adding `-D__BOARDNAME__={build.something}`

bperrybap, 2018-03-23:
I'm not understanding

> we don't add any field not specified in the platform.txt

do to my not understanding how the builder tool works and what boards.txt values it "knows" about.
Are the platform rules only able to access build.xxx variables from the boards.txt file?
If so I understand the issue.

What I was wanting was to have a macro like ARDUINO_BOARD_NAME that was a string based on the existing xxx.name field in the boards.txt file, but it sounds like that may not be possible with the existing code. i.e. I had assumed (perhaps incorrectly)  it was only a simple change to platform.txt

facchinm, 2018-03-26:
In fact you also can use the `name` field, but my point was really about the fact that you only need to add it to the platform.tx, not to the builder as an override.
I just tested adding `'-D__BOARDNAME__="{name}"'` to all recipe patterns and it works just fine.

bperrybap, 2018-03-26:
Yes that is what I meant. I wasn't looking for any kind of override or anything special in builder.
I was looking for the ability of a sketch to get access to the current board name as a string at compile time to allow it to be printed to the user by certain diagnostic sketches.
I was hoping it could be done by a simple platform.txt update which it sounds like it can.
Not sure what the ideal symbol name should be but at least it sounds easy to do, should this feature request be something that is accepted.
__BOARDNAME__ does not seem consistent with the other existing defines used which all begin with ARDUINO