arduino/Arduino #7366: error in dimesion of arduino zero

purplepundit, 2018-03-23:
https://store.arduino.cc/usa/arduino-zero

tech specs has 30 mm width but other place it is 2.1 inch. I am not sure which one is right.

per1234, 2018-03-23:
Note that width is only listed as 30 mm on the Tech Specs tab if you have your location set to America, Asia, Oceania. If you change location to Africa, Europe then it says width is 53 mm.

It seems to be consistently problematic getting the content synced between the product pages for each location.

purplepundit, 2018-03-24:
I am confused now. So if I order in USA I will get 30 mm width board?

Regards,

Sunil Barot

per1234, 2018-03-24:
30 mm is just a typo. You will get a 53 mm wide Zero no matter where you order it. If you want a smaller SAMD board then check out the MKR Zero (or any of the MKR boards). That looks like a nice board.

The reason I added the information about the typo only being on the America/Asia/Oceania version of the product page is that many Arduino employees are in Europe and so they might not see the typo you reported without changing their location on the website.

Thanks for reporting this error purplepundit!

kengdahl, 2018-04-03:
Fixed now

closed 2018-04-03 by kengdahl