arduino/Arduino #7368: In create, provide an "about" menu item to reveal the version being used.

leemke, 2018-03-24:
The title says everything. 
Why?
  because the problem that is reported may be fixed in old releases.

  The problem may not be with Arduino Create, but occurs when using it.
  It would be helpful to add the Create version for people supporting other
  software to know this information. 

  The situation is that software version occur very quickly, it is hard to know 
  if the release you are currently using is the same at the one yesterday.

Just a thought.

per1234, 2018-03-25:
Following official policy established by https://github.com/arduino/arduino-create/issues/1#issuecomment-351344954, I have moved this suggestion to the **Create > Editor** section of the Arduino Forum:
http://forum.arduino.cc/index.php?topic=537354

closed 2018-03-25 by per1234

leemke, 2018-03-25:
thank you.  I have no real understanding of when to use GitHub or Arduino Forum.