arduino/Arduino #7371: Please add maximum_data_size parameter value (RAM size) of board to gcc call  with -D option flag

cassy73, 2018-03-26:
Hi!

Please add a "-D option flag" for the gcc call,
to check/detect the RAM size of MCU.

e.g.: On arduino uno: -DRAMSIZE=2048
maximum_data_size=2048

Best regards

matthijskooijman, 2018-03-26:
IIRC avr-gcc already exposes a `RAMEND` variable (which IIRC is 1 less than the size).

cassy73, 2018-03-26:
THX

OK. I undestand.

BUT:
Some official Arduino's header file is contain RAMSIZE definition, SOME DOESN'T (Arduino Due).

I can calculate RAMSIZE with RAMEND-RAMSTART+1
If these variables declared in MCU specific header file.
But only if the MCU specific header file contents this.
More boards developer forget set these parameters.

Why RAMSIZE declared in MCU specific header, if the compiler config is contain this information too.

Best Regards