arduino/Arduino #7373: Uno/Nano consistently crashes Win7 laptop when lid is closed while plugged in

zrcoutur, 2018-03-26:
Dell Precision 7510, Windows 7 SP1. When either a Nano or Uno is plugged into the laptop (and potentially others), closing the lid will cause the computer to become unresponsive and stuck on a black screen, or immediately shut itself off. The Windows lid close action is set to 'Sleep'. The issue is consistent (not intermittent). Only forcing shutdown and restarting will recover it from this state.

BallscrewBob, 2018-04-04:
Does it do that with anything plugged into the NANO eg sensor etc. If so try it without.
It does sound more like a USB power setting than an issue with the NANO.

facchinm, 2018-04-10:
The Nano is connected to the PC using an FTDI converter (if the board is genuine). Since FTDI is by far the most trusted manufacturer of USB-to-Serial converters I don't think that the fault can be attributed to the board. Have you tried checking the USB behaviour under [Power saving advanced settings](https://www.digitalcitizen.life/how-tweak-advanced-settings-power-plan-windows-7)? 
Closing the issue since I don't think we can do anything about it, reopening if any test demonstrate the opposite :slightly_smiling_face:

closed 2018-04-10 by facchinm