arduino/Arduino #7380: Apt package for raspian is outdated

mannerpots, 2018-03-27:
[This page](https://playground.arduino.cc/Linux/Raspbian) states that:
> Note: This will install an old version of Arduino, and may not support some of the newer boards. You can not download a newer version from the website, because there is no ARM version.

I'm not sure which version it installs by default, but it's older than 1.6.14, because it doesn't have the fix described [here](https://github.com/arduino/Arduino/issues/1981) yet.

However, there is an ARM version available on the website.  I have downloaded it using `wget https://downloads.arduino.cc/arduino-1.8.5-linuxarm.tar.xz`, and it works for verifying and uploading from a headless OS.

Can the apt package for raspbian be updated to a newer version?

per1234, 2018-03-27:
I have fixed the outdated information on that Playground page.

Regarding the outdated package issue, I believe that is related to this:
https://github.com/arduino/Arduino/pull/2703

My feeling is that either Arduino needs to take an official role in maintaining the Linux packages or else the packages should be removed. Yes, it's another job, but if you're going to present a keynote speech at the Embedded Linux Conference titled "Arduino & Linux: A Love Story", well... Even beyond this license issue, it just seems like having 3rd parties making random modifications to the IDE and not keeping the packages updated leads to many more problems than if the Arduino IDE was only available for download from Arduino's website.

mannerpots, 2018-03-28:
Ah, makes sense. I guess that resolves this issue then, unless anyone wants to look into this more.

facchinm, 2018-03-28:
@per1234 I mostly agree with you, but historically it has always been the distribution's maintainer duty to compile and package the supported applications. 
We have a huge dependency problem, mostly due to the fact that the "zero external dependencies" approach started by bundling the JRE helps the beginners but becomes a PITA for repackagers. 

I believe @NicoHood 's effort on AUR packets is the way to go; we could add a wrapper to easily create deb packages in the repo itself to ease rebuilding, but we need some advice from the actual Debian/Ubuntu/Raspbian packagers to do the right thing.

tlk, 2020-02-02:
Raspbian is based on Debian and the Arduino package in Debian is indeed outdated. As far as I can tell work on a new Debian package stalled due to a dependency called "MrBean". 

So I did an experiment to remove MrBean from the Arduino IDE and it seems to work fine without it.

See https://github.com/arduino/Arduino/pull/9700

ThorpeJosh, 2020-06-27:
The 1.0.5 version of Arduino Debian (Ubuntu and Raspbian) package is very outdated but still many recent (third-party) tutorials still instruct installing it through the Debian repository. I support either updating or removing it completely.

What is the best way to use arduino in a CI application that needs to be able to install Arduino on headless Debian machines and then use the arduino cli tools?
Is wget the only way to get a version that supports cli?