arduino/Arduino #7383: New lib : TM1638Lite

danja, 2018-03-28:
https://github.com/danja/TM1638lite

Small(ish) interface library from Arduino to TM1638 display chip/card

Keep up the good work!

per1234, 2018-03-28:
There are currently a couple problems that prevent us from adding your library to the Library Manager index:
The library must be located in the root of the repository. This means that you either need to rename the `TM1638lite` subfolder of the repository to `src` ([1.5 library format](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#source-code)) or move the files from the `TM1638lite` subfolder to the root of the repository (1.0 library format).

After that, you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

When those things are done, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

danja, 2018-03-30:
Ok - my first lib, hopefully got it now. I've renamed subfolder to src, tweaked library.properties and done a corresponding release. Fingers crossed.
Thanks!

danja, 2018-04-01:
I should have said : resubmitting -

https://github.com/danja/TM1638lite

Thanks.

per1234, 2018-04-01:
Yes, it looks fine now. Thanks!

Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that.

cmaglie, 2018-04-03:
done

closed 2018-04-03 by cmaglie