arduino/Arduino #7385: Arduino IDE 1.8.5/1.8.6 hangs by updating libraries

blackbannana, 2018-03-28:
I become a message: you have libs to update, then i try to do it... and by the first one  - i'm waiting undefined amount of time... IDE doesn't response - i can click - cancel, and it is me shown: close window, but this is all - nothing more, hanging...
moreover: i can't install any new library- this same; i have full access to my folders (u+rwx)
Linux Debian 64 bits;
i've been tried to remove all : $HOME folder, ~/.android15 ~/Android ... nothing!

tonylambiris, 2018-07-21:
Scratch that, didn't realize the arduino command was a script specifically setting Java 8 JDK to the path:
```
➜  cat /usr/bin/arduino
#!/bin/sh

export PATH=/usr/lib/jvm/java-8-openjdk/jre/bin/:$PATH
exec /usr/share/arduino/arduino "$@"
```

Running /usr/share/arduino/arduino with Java 9 SDK was able to update the libraries but I did get this output:
```
Picked up JAVA_TOOL_OPTIONS: 
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by processing.app.linux.GTKLookAndFeelFixer (file:/usr/share/arduino/lib/arduino-core.jar) to field com.sun.java.swing.plaf.gtk.GTKLookAndFeel.styleFactory
WARNING: Please consider reporting this to the maintainers of processing.app.linux.GTKLookAndFeelFixer
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```

Thanks.

cmaglie, 2018-07-26:
@blackbannana, did you figure it out?
Some possible issues:
- you need to set a proxy (see on the preferences dialog)
- your antivirus is blocking network access to the IDE

triplehx3, 2019-06-05:
Am seeing exactly the same thing on Manjaro 64bit xfce (Arch based) using IDE 1.8.9.

However I have noticed that if you use xkill and kill the unresponsive ide and then relaunch it sometimes it doesnt freeze in the Library Manager.

I have also seen it happen when using the Board Manager to.

Allso possibly unrelated, I cant use the inbuilt serial console as java errors in the ide whenever the serial monitor recieves data and it doesnt show in console.

I get arround this by using screen and will say that the IDE itself runs fine for me just the freezing on using either the Board Manager or Library Manager and the above mentioned issue with the serial coinsole.

Thanks for all your hard work
Chris