arduino/Arduino #7386: Printf....  Why after 5 year of people asking for it, we can't find a way to add it!!!

trlafleur, 2018-03-29:
In looking back at comments on this, its been a MAJOR request over the years! Yes their are ways to hack it to make it work, but isn't it time just to add it to the IDE.....

More and more people are using the 32 bit processors that have lots of resources... for those who are still using the very small 8 bit processor, add a switch in the preference for it... Make everyone happy!

Thanks for the great work.... but lets move on....

per1234, 2018-03-29:
Closing as duplicate of 
https://github.com/arduino/Arduino/issues/5033
We welcome bug reports and feature requests but in the future please take the time to search for existing open and closed issues and pull requests before submitting an issue:
https://github.com/arduino/Arduino/issues?&q=
If you do find the issue has already been reported or a fix has been submitted, you are welcome to give it a "thumbs up" by clicking on the smiley face icon at the top right to show your support. Please only comment on the issue if you have additional information to add.

closed 2018-03-29 by per1234

PaulStoffregen, 2018-03-29:
The Arduino team decided years ago that this will **not** "Make everyone happy".  They concluded it would make a small but vocal subset of experienced C programmers happy.  But it would make a much larger group of inexperienced novices rather unhappy when trying to understand example programs they find scattered across the internet.

You may disagree.  But don't delude yourself into believing they made this decision lightly, or that it's lingering indecision, or neglectfully ignoring user requests.  It is a very intentional decision the Arduino developers made.  Their reasoning has been explained over and over on the developer mailing list, and now I'm repeating it just for your sake.