arduino/Arduino #7389: Need extra Library paths for Sketch Folder

seachord, 2018-03-29:
I currently have well over a hundred libraries accumulated.  I'm sure many active users have even more!  It getting clumsy to sort thru them by eye!  It would be very helpful if there was a way to have a hierarchical menu system so Libraries could be sorted into folders with meaningful names, and easier to find.

I suspect this might make it harder/slower for the libraries to be found by the IDE. But the the vast wealth of libraries available - it might be worth it!

matthijskooijman, 2018-03-29:
A long time ago I already implemented this, suffering from the same problem. See https://github.com/arduino/Arduino/pull/1986. IIRC that stalled because people could not agree on how to present this in the UI, and by now I believe the PR is probably completely outdated (but might be revamped if there was interest).

lmihalkovic, 2018-04-29:
A better solution to me would be to have all the libs stored inside the sketch to ensure the long term repeatability.. but i realize this is the developer’s approach... not the maker’s.

per1234, 2018-04-29:
@lmihalkovic that's completely off topic for this issue. It's also already possible to do that so no need to discuss it.

lmihalkovic, 2018-04-29:
Nope .. like it or not it is a fact. Which means that the current solution (fat single folder with 100 libs) is not a real issue (even if i wished it would be considered as one)

eagl1, 2019-05-14:
So !! Did you find any solution for it ?