arduino/Arduino #7396: [Library Manager] ShiftRegister74HC595

Simsso, 2018-04-01:
This library has been around for quite a while and is now (with version v1.1.0) compliant with the Library Manager requirements ([v1.1.0](https://github.com/Simsso/ShiftRegister74HC595/releases/tag/v1.1.0)).

This issue is kindly requesting to add the library publicly.

Simsso, 2018-04-01:
Apparently the library was found by the crawler:
[image]
I assume this issue can be close then.

per1234, 2018-04-01:
Library Manager lists libraries you already have installed on your computer as well as those available for installation from the Library Manager index.

Your repository must be added to the Library Manager index before it will be available for installation via Library Manager. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that.

cmaglie, 2018-04-03:
Done

closed 2018-04-03 by cmaglie