arduino/Arduino #7412: cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer

dynek, 2018-04-03:
Hello,

I have been able to upload code onto my ESP8266 for quite a long time and I think (_I think_) one of the last update broke something cause since a few days I'm getting:
```
Archiving built core (caching) in: /var/folders/px/5d3hxcpd2jz0j2bfvsv7mlzc0000gn/T/arduino_cache_939222/core/core_esp8266_esp8266_generic_CpuFrequency_80,ResetMethod_ck,CrystalFreq_26,FlashFreq_40,FlashMode_qio,FlashSize_512K0,led_2,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_all,UploadSpeed_115200_ba49459f356d8c9f9eec109967582725.a
Sketch uses 263040 bytes (52%) of program storage space. Maximum is 499696 bytes.
Global variables use 33956 bytes (41%) of dynamic memory, leaving 47964 bytes for local variables. Maximum is 81920 bytes.
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer(SerialUploader.java:314)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:89)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:713)
	at processing.app.SketchController.exportApplet(SketchController.java:686)
	at processing.app.Editor$DefaultExportAppHandler.run(Editor.java:2265)
	at java.lang.Thread.run(Thread.java:748)
```
I am able to use esptool on command line to communicate with the ESP:
```
$ esptool.py --port /dev/cu.SLAB_USBtoUART erase_flash
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 3.0s
Hard resetting...
```
**Versions**
Arduino IDE: 1.8.5
esp8266 board package: 2.4.1

Google has not been able to help this time :-( Any hint will be much appreciated.

Thanks in advance!

per1234, 2018-04-03:
I get this when attempting an "Upload Using Programmer" to an ESP8266 board, which is the expected result since that operation is not supported by the ESP8266 core for Arduino. Are you holding the shift key while clicking the upload button, using the Ctrl + Shift + U keyboard shortcut, or **Sketch > Upload Using Programmer** when uploading?

dynek, 2018-04-04:
Not sure how I ended up pressing **shift** but yes that was the problem.
Think it might have worked in the past when uploading to esp8266 was still a programmer?
Thanks for pointing that out.

closed 2018-04-04 by dynek

mbenothman, 2019-08-27:
Hi
Could help explaining what is the difference between Ctrl + Shift + U keyboard shortcut AND Sketch > Upload Using Programmer when uploading? It happens I'm facing the same java error.