arduino/Arduino #7416: Add Google Cloud IoT Core JWT library

gguuss, 2018-04-04:
For the past few months we (myself and @cirvladimir) have been working on getting some Arduino support for Google Cloud IoT Core. I have, to the best of my knowledge, produced a library.properties file that confirms to the Arduino Library 1.5 specification and have tested the library locally on my machine.

We have published our library source code here:

* https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino

And I tagged a release for initial inclusion here:

* https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/releases/tag/v1.0.0-alpha1

Let me know if there's anything else I can do to get us included in the library manager!

gguuss, 2018-04-09:
Done, removed / from name, also added support for Genuino boards. 

New release pushed here:
    https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/releases/tag/1.0.1-alpha

gguuss, 2018-04-09:
And @cmaglie, thanks for the update, I have deleted the offending release.

per1234, 2018-04-11:
@yi15 this is not an appropriate place to request support.

yi15, 2018-04-11:
Sorry. May I know where can I request for support? @per1234

gguuss, 2018-04-12:
@yi15 this issue you're commenting on is for requesting inclusion in the Arduino Library manager for our library but is not support for it. [File an issue on the relevant project GitHub](https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino) and we'll be there to help you. Please refrain from further off-topic comments here.

yi15, 2018-04-12:
ok, thank you @gguuss