arduino/Arduino #7419: Please add library Timing

elhayra, 2018-04-05:
github: https://github.com/elhayra/timing
release: https://github.com/elhayra/timing/releases/tag/v1.0

Thanks!

per1234, 2018-04-05:
Your repository must contain a library.properties file. The format is documented here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

After adding that file please create a new tag or release that matches the `version` value set in your library.properties file.

Once that's complete, please comment here so we know we can continue with adding your library to the Library Manager index.

elhayra, 2018-04-07:
I decided to split the library into small libraries, since each class is meant to do different type of work.
I added library.properties to each as you requested.
Those are the links for the libraries:
Strober:
github: https://github.com/elhayra/Strober
release: https://github.com/elhayra/Strober/releases/tag/v1.0.0

SandTimer:
github: https://github.com/elhayra/SandTimer
release: https://github.com/elhayra/SandTimer/releases/tag/V1.0.0

MilliStopper:
github: https://github.com/elhayra/MilliStopper
release: https://github.com/elhayra/MilliStopper/releases/tag/v1.0.0

per1234, 2018-04-07:
There's an issue with your Strober library: The [library.properties `name` field](https://github.com/elhayra/Strober/blob/master/library.properties#L1) is set to SandTimer, which will conflict with the name of your SandTimer library. Please update the file (note the `url` and `includes` fields also have the SandTimer info) and then redo the tag or create a new one, then comment here so we know we can proceed.

elhayra, 2018-04-08:
Sorry for that. Must be a copy-paste error.
I fixed it, and created a new tag:
https://github.com/elhayra/Strober/releases/tag/V1.0.1

cmaglie, 2018-04-09:
Done

closed 2018-04-09 by cmaglie

elhayra, 2018-04-14:
Where can I see my library on Arduino website? I looked [here](https://playground.arduino.cc/Main/LibraryList#Timing) and coudn't find it

per1234, 2018-04-15:
@elhayra your request was interpreted as you wanting your libraries to be added to the Arduino Library Manager index. That allows people to easily find/install/update your libraries via the Arduino IDE's **Sketch > Include Library > Manage Libraries**. More information here:
https://www.arduino.cc/en/Guide/Libraries#toc3

Your library won't ever appear on the Library List page on the Arduino website. That is not a service Arduino offers for 3rd party libraries. There are some contributed libraries listed on that page but they are [terribly outdated](https://github.com/arduino/Arduino/issues/5291#issuecomment-242618294) and need to be removed, rather than adding more.

It is in the Library Manager index JSON file:
http://downloads.arduino.cc/libraries/library_index.json
You are welcome to add it to the library list on the Arduino Playground if you like:
https://playground.arduino.cc/Main/LibraryList
The Arduino Playground is a publicly editable wiki. You just need to sign in/sign up for your Arduino account and then you will see an "edit" link at the top of that page. When you click the "edit" link you will arrive at a password input page but the password is pre-filled so you only need to do the captcha and then click the "Submit" button and you'll arrive at the editing page.