arduino/Arduino #7420: THREE New GSM Shields v2 - GSM Library Scrap

Thor2018, 2018-04-05:
Hello everyone,

I have lot's of problems with three new original Arduino GSM Shields v2. Everyone does not functioning correctly because the **GSM Library is incorrect**. You don't belive me?
Google GSM Shield Problems! I can't understand why the hell you **sell products which don't function** correctly! I have spent more than 300$ on, so it seems, not functioning stuff!

Also the **Arduino Forum can't help me!**

After i tested one of the original Arduino GSM Shields v2 with the Original Mega2560, everything worked fine (Sending and Receiving SMS).
But today suddenly the GSM Modul stucks at the command "ReceiveSMS" or "GsmScanNetworks" from the Example Sketches!

I already tried and have done:

-     Connect Pin 2 from GSM Shield v2 to Pin 10 of Arduino Mega 2560 R3
-     My power supply can provide up to 3A by 12V
-     Removed the PIN from the SIM Card
-     The SIM Card itself is working
-     The Power Button on the GSM Shield is ON
-     Another New GSM Shield, for trying out, has the same problem
-     Another New Mega2560 comes up with the same problem

I also tried out something with AT commands like the Sketch below. Then i get after i typed in **AT** the following answer

[image]

So do you know any further steps to can solve this problem? I can't understand why they selling products (GSM Shield's) which are not working. Especially they are really expensive!

I'm thankful for every answer!

Best regards, Thorsten

_`
#include <SoftwareSerial.h>
#include <string.h>

char incoming_char = 0;
SoftwareSerial cell(10,3);

void setup()
{
  // Initialize serial ports for communication.
  Serial.begin(9600);

  // Reset
  Serial.println("Start Reset");
  pinMode(7, OUTPUT);
  digitalWrite(7, HIGH);
  delay(12000);
  digitalWrite(7, LOW);
  delay(1000);
  Serial.println("End Reset");

  cell.begin(9600);

  Serial.println("Enter your AT commands (with CR & NL)...");
}

void loop()
{
  if(cell.available() > 0)
  {
    incoming_char = cell.read();
    if((incoming_char >= ' ') && (incoming_char<='z'))
      Serial.print(incoming_char);
    else
    {
      Serial.print("%");
      Serial.print((int) incoming_char);
      Serial.print("%");
      if(incoming_char == 10)
        Serial.println();
    }
  }

  if(Serial.available() > 0)
  {
    incoming_char = Serial.read();
    cell.print(incoming_char);
  }
}
`_

dlabun, 2018-04-07:
I should first point out that the GSM Shield v2 has been retired for some time now and Arduino isn't selling them any more...

Are you saying you tried the testing sketch here:
[https://www.arduino.cc/en/Guide/ArduinoGSMShieldV2#toc7](url)
 
and that's failing? If so, you may have an issue with the SIM itself not being permitted onto the network. The modem may be reporting back network failure codes the library isn't capable of handling. The best way to troubleshoot the modem is to connect the modem to a USB to UART cable and use a terminal program on your PC so send the AT commands.

per1234, 2021-03-28:
Transferred to https://github.com/arduino-libraries/GSM/issues/20

closed 2021-03-28 by per1234